nopnop2002/SC16IS752 #9: I2c address 0x4D

Listian, 2020-01-06:
Hello! I have the same module. The I2c scanner transmits the address 0x4D. When applying vdd or gnd to A0 / A1, the address does not change. And the example with 0x4D does not work. Do not know what the problem is?

nopnop2002, 2020-01-06:
Is your environment Arduino??
Is Rpi??

Listian, 2020-01-06:
Sorry, forgot to specify the platform and IDE. 

ESP 32 lolin32, Arduino IDE. I use the code from your example.

In the code library found device_address_sspin = (addr_sspin >> 1);

Removing the shift anyway at 0x4D does not work.

Log Code:
12: 27: 44.968 -> ReadRegister channel = 0 reg_addr = 70 result = FF
12: 27: 45.001 -> WriteRegister channel = 0 reg_addr = 70 val = FF
12: 27: 45.034 -> ReadRegister channel = 0 reg_addr = 10 result = FF
12: 27: 45.102 -> WriteRegister channel = 0 reg_addr = 10 val = FF
12: 27: 45.135 -> ReadRegister channel = 1 reg_addr = 12 result = FF
12: 27: 45.202 -> WriteRegister channel = 1 reg_addr = 12 val = FF
12: 27: 45.235 -> ReadRegister channel = 0 reg_addr = 20 result = FF
12: 27: 45.302 -> ReadRegister channel = 0 reg_addr = 18 result = FF
12: 27: 45.335 -> WriteRegister channel = 0 reg_addr = 18 val = FF
12: 27: 45.368 -> WriteRegister channel = 0 reg_addr = 0 val = 3
12: 27: 45.435 -> WriteRegister channel = 0 reg_addr = 8 val = 0
12: 27: 45.468 -> WriteRegister channel = 0 reg_addr = 18 val = 7F
12: 27: 45.501 -> Desired baudrate: 9600
12: 27: 45.535 -> Calculated divisor: 3
12: 27: 45.568 -> Actual baudrate: 9600
12: 27: 45.601 -> Baudrate error: 0
12: 27: 45.601 -> ReadRegister channel = 1 reg_addr = 22 result = FF
12: 27: 45.669 -> ReadRegister channel = 1 reg_addr = 1A result = FF
12: 27: 45.702 -> WriteRegister channel = 1 reg_addr = 1A val = FF
12: 27: 45.736 -> WriteRegister channel = 1 reg_addr = 2 val = 3
12: 27: 45.803 -> WriteRegister channel = 1 reg_addr = A val = 0
12: 27: 45.836 -> WriteRegister channel = 1 reg_addr = 1A val = 7F
12: 27: 45.869 -> Desired baudrate: 9600
12: 27: 45.903 -> Calculated divisor: 3
12: 27: 45.936 -> Actual baudrate: 9600
12: 27: 45.969 -> Baudrate error: 0
12: 27: 45.969 -> ReadRegister channel = 0 reg_addr = 18 result = FF
12: 27: 46.036 -> LCR Register: 0x192
12: 27: 46.036 -> WriteRegister channel = 0 reg_addr = 18 val = C3
12: 27: 46.103 -> ReadRegister channel = 1 reg_addr = 1A result = FF
12: 27: 46.136 -> LCR Register: 0x192
12: 27: 46.171 -> WriteRegister channel = 1 reg_addr = 1A val = C3
12: 27: 46.204 -> WriteRegister channel = 0 reg_addr = 38 val = 55
12: 27: 46.237 -> ReadRegister channel = 0 reg_addr = 38 result = FF
12: 27: 46.305 -> Device not found
--------

I wrote a little code and it works at 0x4D. Detects pin when connected to ground.

#include <Wire.h>

void setup() {
  Serial.begin(9600);
  Wire.begin();  
}
void loop() {

  uint8_t reg_addr = 0x0B;
  uint8_t channel = 0x00;
  Wire.beginTransmission(0x4D); 
 Wire.write((reg_addr<<3 | channel<<1)); 
    Wire.endTransmission(0); 
    Wire.requestFrom(0x4D, 1);          
  uint8_t c = Wire.read(); 
   Serial.println(c, HEX);
  delay(1000);
}

nopnop2002, 2020-01-06:
The i2c address is based on the following table.

https://github.com/nopnop2002/SC16IS752/blob/master/Arduino%20%26%20ESP8266/SC16IS752.h#L38

I2c scanner transmits the address is 0x4D:

SC16IS752 i2cuart = SC16IS752(SC16IS750_PROTOCOL_I2C,SC16IS750_ADDRESS_BB);

Listian, 2020-01-07:
I still do not understand why your library sends 0xFF in response.

I studied your library and module documentation. And using it, I managed to record and receive data from the registers at the address 0x4D.

In the documentation
 when working with ESR you need to write 0xBF to the LCR. Studying your code, I never found a translation into this mode. In doing so, you are trying to work with ESR O.o

As I understand it, the microcircuit has its own addresses, and the module uses another (with a shift to the right).
This should have been indicated in the examples. But I still did not work.

There will be time to see why your library does not work for me.

But the main thing is that the module itself still works! )))

nopnop2002, 2020-01-08:
ESP32 has three UART port as standard.
Do you need more UARTs? ?

Listian, 2020-01-12:
I already wrote that the module is working)
It works also from 3.3V, but yes it lowers all output voltages on the module.

p.s.
I took the time and looked at your library why it does not work for me.

Everything is very simple, except for the problem with the I2C address, the problem is still this:

#ifdef __AVR__
 #define WIRE Wire
#elif ESP8266 // ESP8266
 #define WIRE Wire
#else // Arduino Due
 #define WIRE Wire1
#endif

Really very strange announcement of the WIRE constant for different microcontrollers%)

Because of this, it did not work for me with ESP32.

In general, the question is closed) I will write my library for my projects. Yours is still very raw. There is still a problem with ESR registers.

Good luck with the past holidays!)

nopnop2002, 2020-01-13:
Thank you for your report.

closed 2020-01-13 by nopnop2002