nopnop2002/SC16IS752 #10: Using interrupts on digital pins

szerwi, 2021-02-10:
Hi,
Can someone explain me how to use interrupts in this expander? I would like to detect changes on digital pins. Does the expander store pin number by which the interrupt was trigerred or do I need to read all the pins and define which pin trigerred the interrupt by myself in Arduino code?

nopnop2002, 2021-02-10:
There is dataheet.

https://www.nxp.com/docs/en/data-sheet/SC16IS752_SC16IS762.pdf

There is an explanation about I/O Interrupt Enable register in Chapter 8.17.

szerwi, 2021-02-10:
@nopnop2002 I have read the datasheet, but I don't understand it perfectly. I'm not sure if these registers stor the information about the pin which trigerred the interrupt or not.

nopnop2002, 2021-12-30:
I tested GPIO input interrupts.

By setting the register, the value of the IIR register at the time of input changes.

However, the value of the IRQ pin does not change.

It monitors the IIR register.

```
i2c_addressr=4d
device found
temp_ier=0
temp_ier=8
temp_ioint=0
mask=4
temp_ioint=4
temp_iir=c1
temp_iir=c1
temp_iir=c1
temp_iir=f0 ---> input pin change of state. But IRQ signal is no change.
temp_iir=c1
temp_iir=c1
temp_iir=f0 ---> input pin change of state. But IRQ signal is no change.
temp_iir=c1
temp_iir=c1

```

nopnop2002, 2022-01-01:
I found a reason.

There is a bug at the PCB routing:
All boards are missing the wire between the IRQ pin of the SC16IS750 and the IRQ connector at the board.

szerwi, 2022-01-01:
Thanks. Shortly after asking this question I've also managed to get the interrupts working.

closed 2022-01-01 by szerwi