nopnop2002/SC16IS752 #11: RS485 ESP32 half-duplex communication

peter1a, 2021-12-20:
Hi, I have a question, so I have two identical devices that I would like to communicate via RX / TX (RS485). Both devices are exactly the same component, but as far as I know this is half-duplex mode, which means the first device sends, the second device receives and then again the second device receives and the first device sends. I have no idea how to write code for this communication. I found here as far as this chip is concerned GitHub - nopnop2002/SC16IS752: Driver for SC16IS752 also tried to load on both devices example code SC16IS752/Arduino & ESP8266/example/I2C_RECEIVE at master · nopnop2002/SC16IS752 · GitHub but nothing works. Some ideas?

nopnop2002, 2021-12-20:
RS485 communication is not possible with this module.

nopnop2002, 2021-12-20:
Or you need such a UART and R485 conversion module
https://www.aliexpress.com/item/1005001632287571.html?spm=a2g0o.productlist.0.0.65b46866jIu2y6&algo_pvid=21019853-1e5c-445a-8571-1bbd388b680f&algo_exp_id=21019853-1e5c-445a-8571-1bbd388b680f-6&pdp_ext_f=%7B%22sku_id%22%3A%2212000016899329124%22%7D

peter1a, 2021-12-21:
I have this example code (one for sending / one for receiving) is that OK? 
`#include <SPI.h>
#include <Wire.h>
#include <SC16IS752.h>
# define BUZZER 1

SC16IS752 i2cuart = SC16IS752(SC16IS750_PROTOCOL_I2C, 0x48);

#define baudrate_A 9600
#define baudrate_B 9600

String type = "w";

void setup() {
  Serial.begin(115200);
  pinMode(21, OUTPUT);
  digitalWrite(21, HIGH);

  i2cuart.begin(baudrate_A, baudrate_B);
  if (i2cuart.ping() != 1) {
    Serial.println("device not found");
    while (1);
  } else {
    Serial.println("device found");
  }

  i2cuart.pinMode(BUZZER, OUTPUT);
}

void loop() {
  if (type == "w") {
    Serial.println("writing");

    // WRITE
    i2cuart.write(SC16IS752_CHANNEL_B, 123);

    delay(1000);
  } else {
    Serial.println("reading");

    // READ
    if (i2cuart.available(SC16IS752_CHANNEL_B) > 0) {
      alertBuzzer();

      // read the incoming byte:
      char c = i2cuart.read(SC16IS752_CHANNEL_B);

      Serial.println(c);

      delay(1000);
    }
  }
}

void alertBuzzer() {
  for (int i = 0; i <= 255; i++) {
    i2cuart.digitalWrite(BUZZER, HIGH);
    delay(1);
    i2cuart.digitalWrite(BUZZER, LOW);
  }
}`

nopnop2002, 2021-12-21:
This is better.

```
// READ
if (i2cuart.available(SC16IS752_CHANNEL_B) > 0) {

  // read the incoming byte:
  char c = i2cuart.read(SC16IS752_CHANNEL_B);

  Serial.println(c);

  alertBuzzer();

  delay(1000);
}

peter1a, 2021-12-21:
So if I use one channel for sending this should be also for receiving? For example if I send to CHANNEL A from device 1 should be on device 2 CHANNEL A as reader?

nopnop2002, 2021-12-21:
Channel A and channel B are completely independent.

peter1a, 2021-12-21:
Should I call somewhere function EnableTransmit() ?

nopnop2002, 2021-12-21:
>Should I call somewhere function EnableTransmit() ?

Yes.

peter1a, 2021-12-21:
Can you provide me more info about how to call and where, I try this: i2cuart.EnableTransmit(SC16IS752_CHANNEL_A); but doesn't work.

nopnop2002, 2021-12-21:
>Should I call somewhere function EnableTransmit() ?

Sorry. 
I was wrong.
No.

peter1a, 2021-12-21:
This won't work because it's half-duplex one another out not at same time. I also change EnableTransmit but doesn't work neither.

nopnop2002, 2021-12-21:
Can you provide the timing chart of sending and receiving that you want to do?

peter1a, 2021-12-21:
I just want to send from one device to another and than back, so for example lets says we have in my case 2 same device with same components. Device A send some data (plain text for example) to device B, on receive device B use buzzer so I know data was received (currently just for debbuging process). Currently I just want to have this in work so from device A send data and receive on device B. I made it work on two ESP32 direclty via RX/TX pins but don't know how is in this case when I have extra components on board  SC16IS752 (double UART) and SN65HVD72 (RS485).

nopnop2002, 2021-12-21:
Is this what you want to do?

Device A -----> Device B
Device B call buzzer.

peter1a, 2021-12-21:
Correct.

nopnop2002, 2021-12-21:
Where is the ESP32?

peter1a, 2021-12-21:
Sorry forgot to post schematic, here you go https://ibb.co/dr8zf5M.

nopnop2002, 2021-12-21:
Device ----- (RS485) ----> SN65HVD72D
SN65HVD72D ---- (UART) ----> SC16IS752IPW
SC16IS752IPW --- (I2C) -----> ESP32
ESP32 --- (SC16IS752IPW) ---> GPIO1

Is this what you want to do?

peter1a, 2021-12-21:
Correct.

nopnop2002, 2021-12-21:
Why use SC16IS752IPW?

Device ----- (RS485) ----> SN65HVD72D
SN65HVD72D ---- (UART) ----> ESP32
ESP32 --- --> GPIO1

peter1a, 2021-12-21:
We already have developed PCBs so I can't change it, I need to use what I have...

nopnop2002, 2021-12-21:
Not tested.
```
#include <SPI.h>
#include <Wire.h>
#include <SC16IS752.h>

SC16IS752 i2cuart = SC16IS752(SC16IS750_PROTOCOL_I2C, 0x48);

#define baudrate_A 9600
#define baudrate_B 9600
#define GPIO 1

String type = "w";

void setup() {
	i2cuart.begin(baudrate_A, baudrate_B);
	if (i2cuart.ping() != 1) {
		Serial.println("device not found");
		while (1);
	}
	Serial.println("device found");
    i2cuart.pinMode(GPIO, OUTPUT);
    i2cuart.digitalWrite(GPIO, LOW);
}

void loop() {
	if (i2cuart.available(SC16IS752_CHANNEL_A) > 0) {
		// read the incoming byte:
		char c = i2cuart.read(SC16IS752_CHANNEL_A);
		i2cuart.digitalWrite(GPIO, HIGH);
		delay(1000);
		i2cuart.digitalWrite(GPIO, LOW);
		delay(1000);
	}
}
```

peter1a, 2021-12-21:
What should I do on write site?

nopnop2002, 2021-12-21:
What is "write site" ?

peter1a, 2021-12-21:
I mean on device from where I send data (device A)?

nopnop2002, 2021-12-21:
You send data using RS485.

peter1a, 2021-12-21:
Ok for example if I use just ESP32 without any components as device A, I send data to this device via RX / TX as that: 
`void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.write("writing");
  delay(1000);
}`

On device B I should upload your sketch witch would beep on each receive of data from TX/RX right?

nopnop2002, 2021-12-21:
Your PCB always requires a device other than the ESP32.

Serial.write ("writing") just prints the characters on the screen.

In my sketch, it BEEPs every time it receives from RS485.

peter1a, 2021-12-21:
Aha OK so I need some device which have RS485 to send data, right?

peter1a, 2021-12-21:
Also what if I have two same devices which have same components as on schematic and from one I send from another I receive, as both have RS485 and all other things?

nopnop2002, 2021-12-21:
It good idea.

The sender and receiver need to perform different sketches.

peter1a, 2021-12-21:
Yes I need another sketch for sender in this case, but don't know how to write sketch for sending data in this case, can you help me with that?

nopnop2002, 2021-12-21:
```
#include <SPI.h>
#include <Wire.h>
#include <SC16IS752.h>

SC16IS752 i2cuart = SC16IS752(SC16IS750_PROTOCOL_I2C, 0x48);

#define baudrate_A 9600
#define baudrate_B 9600

void setup() {
	Serial.begin(115200);
	i2cuart.begin(baudrate_A, baudrate_B);
	if (i2cuart.ping() != 1) {
		Serial.println("device not found");
		while (1);
	}
	Serial.println("device found");
}

void loop() {
  i2cuart.write(SC16IS752_CHANNEL_A, 123);
  delay(1000);
}
```

nopnop2002, 2021-12-21:
If it doesn't work, you need another RS485 transmitter circuit.

There may be a problem with the PCB board.

peter1a, 2021-12-21:
Nop it doesn't working...

peter1a, 2021-12-21:
What if there should be some changes on your driver for supporting RS485? Or is this already included?

nopnop2002, 2021-12-21:
If you use your PCB, you have nothing.
If you don't use your PCB, this is better way.

Device(Not ESP32) ----- (RS485) ----> SN65HVD72D
SN65HVD72D ---- (UART) ----> SC16IS752IPW
SC16IS752IPW --- (I2C) -----> ESP32
ESP32 --- (SC16IS752IPW) ---> GPIO1

nopnop2002, 2021-12-21:
I have never used SN65HVD72D.
But Your PCB circuit may be wrong.

Check this.

https://www.mikrocontroller.net/attachment/485007/Screenshot_2020-12-17_094541.png

peter1a, 2021-12-21:
What if I go in oposite direction, if I use device and send data with   i2cuart.write(SC16IS752_CHANNEL_A, 123);, should I receive it on ESP32 RX pin directly? So I could check that data has been send right via RS485?

nopnop2002, 2021-12-21:
It doesn't make sense to test with a PCB as it is unclear if the PCB circuit is correct.

peter1a, 2021-12-22:
After hours of reading I understand I need to set pin to low/high when receiving or transmiting. So I need to set pins (basically one pin for both on RS485 component DE & !RE) to high/low. Those pin is in my case connected to RTSA pin of UART which should do this job automatically based on datasheet, do I need to enable somehow auto RTS? In datasheet it says:
This pin only affects the transmit and receive operations when Auto-RTS function is enabled via the Enhanced Features Register (EFR[6]) for hardware flow control operation.

Would be better in my case if I use some of free GPIO pins from UART and manually changing pin to low/high?

nopnop2002, 2021-12-22:
>I need to set pin to low/high when receiving or transmiting. So I need to set pins (basically one pin for both on RS485 component DE & !RE) to high/low. 

Yes. 
That's right.

>This pin only affects the transmit and receive operations when Auto-RTS function is enabled via the Enhanced Features Register (EFR[6]) for hardware flow control operation.

I think that DE and / RE can be automatically switched to High / Low by properly setting one of the SC16IS752IPW registers, but I'm not sure.

>Would be better in my case if I use some of free GPIO pins from UART and manually changing pin to low/high?

Yes.
That is the better way.

nopnop2002, 2021-12-22:
From datasheet.

https://www.nxp.com/docs/en/data-sheet/SC16IS752_SC16IS762.pdf

```
9.1 Auto RS-485 RTS control

Normally the RTS pin is controlled by MCR bit 1, or if hardware flow control is enabled, the logic state of the RTS pin is controlled by the hardware flow control circuitry. 

EFCR register bit 4 will take the precedence over the other two modes; 

once this bit is set, the transmitter will control the state of the RTS pin. 

The transmitter automatically asserts the RTS pin (logic 0) once the host writes data to the transmit FIFO, and de-asserts RTS pin (logic 1) once the last bit of the data has been transmitted.

To use the auto RS-485 RTS mode the software would have to disable the hardware flow control function.
```

It may be possible to automatically switch DE and / RE to High / Low by setting Bit4 in the EFCR register.
__But I haven't confirmed.__

This function changes the EFCR register.
https://github.com/nopnop2002/SC16IS752/blob/master/Arduino%20%26%20ESP8266/SC16IS752.cpp#L503

peter1a, 2021-12-22:
Yes I was also looking this functions, but strage this functions are never caled, I try with Serial.println() inside this function but no output so I don't know if those are only writen but never caled?

nopnop2002, 2021-12-22:
>I don't know if those are only writen but never called?

EnableTransmit function never called.

It is only defined as a function.

I have never used this function.

peter1a, 2021-12-22:
From where is this library original? Maybe I can find how those functions were called...

nopnop2002, 2021-12-22:
Original library:
https://github.com/SandboxElectronics/UART_Bridge

This is for SC16IS750.

nopnop2002, 2021-12-22:
You need to call EnableTransmit () in your sketch,

nopnop2002, 2021-12-22:
DE and RE are interconnected and can be transmitted by setting free GPIO to HIGH at the time of transmission.
It may be necessary to set bit 5 of the EFCR register.
But there is no such function in my library.

RTSINVER Invert RTS signal in RS-485 mode.
0: RTS = 0 during transmission and RTS = 1 during reception
1: RTS = 1 during transmission and RTS = 0 during reception

peter1a, 2021-12-22:
EnableTransmit is private so it should be called inside library, but don't know where to put it...

nopnop2002, 2021-12-22:
```void SC16IS752::begin(uint32_t baud_A, uint32_t baud_B)
{
	if ( protocol == SC16IS750_PROTOCOL_I2C) {
		WIRE.begin();
	} else {
		::pinMode(device_address_sspin, OUTPUT);
		::digitalWrite(device_address_sspin, HIGH);
		SPI.setDataMode(SPI_MODE0);
		SPI.setClockDivider(SPI_CLOCK_DIV4);
		SPI.setBitOrder(MSBFIRST);
		SPI.begin();
		//SPI.setClockDivider(32);
	};
	ResetDevice();
	FIFOEnable(SC16IS752_CHANNEL_A, 1);
	FIFOEnable(SC16IS752_CHANNEL_B, 1);
	SetBaudrate(SC16IS752_CHANNEL_A, baud_A);
	SetBaudrate(SC16IS752_CHANNEL_B, baud_B);
	SetLine(SC16IS752_CHANNEL_A, 8, 0, 1);
	SetLine(SC16IS752_CHANNEL_B, 8, 0, 1);

       EnableTransmit(SC16IS752_CHANNEL_A, 0);
       EnableTransmit(SC16IS752_CHANNEL_B, 0);
}
```

It may be necessary to set bit 5 of the EFCR register of the SC16IS752I.
But there is no such function in my library.

RTSINVER Invert RTS signal in RS-485 mode.
0: RTS = 0 during transmission and RTS = 1 during reception
1: RTS = 1 during transmission and RTS = 0 during reception

peter1a, 2021-12-22:
Noup, not working...

nopnop2002, 2021-12-22:
EDIT

```
void SC16IS752::begin(uint32_t baud_A, uint32_t baud_B)
{
	if ( protocol == SC16IS750_PROTOCOL_I2C) {
		WIRE.begin();
	} else {
		::pinMode(device_address_sspin, OUTPUT);
		::digitalWrite(device_address_sspin, HIGH);
		SPI.setDataMode(SPI_MODE0);
		SPI.setClockDivider(SPI_CLOCK_DIV4);
		SPI.setBitOrder(MSBFIRST);
		SPI.begin();
		//SPI.setClockDivider(32);
	};
	ResetDevice();
	FIFOEnable(SC16IS752_CHANNEL_A, 1);
	FIFOEnable(SC16IS752_CHANNEL_B, 1);
	SetBaudrate(SC16IS752_CHANNEL_A, baud_A);
	SetBaudrate(SC16IS752_CHANNEL_B, baud_B);
	SetLine(SC16IS752_CHANNEL_A, 8, 0, 1);
	SetLine(SC16IS752_CHANNEL_B, 8, 0, 1);

	uint8_t temp_efcr;
	temp_efcr = ReadRegister(SC16IS752_CHANNEL_A, SC16IS750_REG_EFCR);
        temp_efcr |= 0x30
	WriteRegister(SC16IS752_CHANNEL_A, SC16IS750_REG_EFCR,temp_efcr);
}
```

peter1a, 2021-12-22:
Noup, still nothing, do you maybe have mail / skype or something, would be easier to contact each other?

nopnop2002, 2021-12-22:
There is nothong to try.

I bought MX3485.
It will take a month to get it.
I will test it when it arrives.

Manually change DE and /RE to low / high using some of the free GPIO pins on the SC16IS752 or ESP32.
That's a better way.

peter1a, 2021-12-22:
There is no option to set LOW / HIGH manualy RTSA pin?

nopnop2002, 2021-12-22:
I don't have.

peter1a, 2021-12-22:
I mean to set LOW/HIGH manually via register?

nopnop2002, 2021-12-22:
There doesn't seem to be a register to manually switch the RTS signal on the SC16IS762.

If you find, please tell me.

https://www.nxp.com/docs/en/data-sheet/SC16IS752_SC16IS762.pdf

peter1a, 2021-12-22:
I found it here:

[image]

[55 more replies]

closed 2022-03-07 by peter1a