nopnop2002/SC16IS752 #13: How to specifying SCL and SDA.

nopnop2002, 2022-02-13:
esp32:
Wire.setPins(int sdaPin, int sclPin);
i2cuart.begin(baudrate_A, baudrate_B);

esp8266:
Nothing.

atmega:
#define SDA_PORT PORTD
#define SDA_PIN SDA
#define SCL_PORT PORTD
#define SCL_PIN SCL
#include <SoftI2CMaster.h>
#include <SoftWire.h>
SoftWire Wire = SoftWire();
i2cuart.begin(baudrate_A, baudrate_B);

stm32:
Wire.setSCL(SCL);
Wire.setSDA(SDA);
i2cuart.begin(baudrate_A, baudrate_B);

nRF52832:
Wire.setPins(int sdaPin, int sclPin);
i2cuart.begin(baudrate_A, baudrate_B);

nopnop2002, 2022-02-13:
>Wire.setPins(int sdaPin, int sclPin); not working on ESP32 (error while compiling: 'class TwoWire' has no member named 'setPins').

It's not a library issue.
Check here.
https://github.com/espressif/arduino-esp32/issues

mysticrnd, 2022-09-29:
Is this library supports SC16IS762 ?

nopnop2002, 2022-09-29:
@mysticrnd 

The SC16IS762 differs from the SC16IS752 in that it supports SPI clock speeds up to 15 Mbit/s instead of the 4 Mbit/s supported by the SC16IS752, and in that it supports IrDA SIR up to 1.152 Mbit/s.

It probably works.